kachurovskiy/nanoels #7: Adaption for Mach4

Charly333, 2021-05-16:
Hi Maxim, 

what a hell of great ideas! 

I´ve just ordered my lathe and will adapt it to Mach4 as I did with my mill. 

Those motor holders of your´s are well designed, but i´m quite familiar with that heated up nema steppers on plastic holders problem ;) Milled some aluminum holders for my 3d printer to get rid of all this ongoing deformations not only from temperature but from pressure and load change. So for the lathe too, there will be aluminium ones. Therefore I´ll need to make some adaptions on your sketches to make it millable and maybe suitable for closed circuit steppers with different form factors. Could you hand over those sketches as f3d files too? Only found STL.

One off the main problems left will be to adapt the rotary encoder to my PoKeys57CNC board. Any advice in this? You´ve realized a degree readout for the chuck - what is the smallest subdevision you can get out of this?

Second open question is to have chuck rpm programmable. Any idea if motor is adaptable to a normal VFD Inverter that´s triggered buy 0 to 10V?

Maybe a closed loop servo to drive the chuck might be the better solution for those two open questions - how you feel about it?

Reg. Charly

kachurovskiy, 2021-05-16:
Hi Charly, thank you.

> heated up nema steppers on plastic holders problem

Yes, plastic has it's limitations. Have you considered PETG? It has a wider temperature range than PLA and is said to creep less with time.

> Could you hand over those sketches as f3d files too? Only found STL.

Sure, which models exactly are you interested in?

> One off the main problems left will be to adapt the rotary encoder to my PoKeys57CNC board. Any advice in this?

Not sure what you mean here. I assume you'd need to mechanically connect the rotary encoder to the spindle and electronically connect it to the board. I've never worked with that board though.

> Second open question is to have chuck rpm programmable. Any idea if motor is adaptable to a normal VFD Inverter that´s triggered buy 0 to 10V?

Sorry, I've never worked with that, NanoEls doesn't control the spindle motor - only the lead screw.

> Maybe a closed loop servo to drive the chuck might be the better solution for those two open questions - how you feel about it?

Closed loop servo motors/drivers are expensive and the motor might not fit in the enclosure of the default DC motor that most of Chinese lathes come with. Other than looking them up on AliExpress I have no experience with them though :)

Best wishes!

Charly333, 2021-05-16:
Hi, thank´s for your quick response.

Did try a lot, different plastics. PETG is bit better, but I´ve 2,8 nm on y axis of printer (4 colors with 4 separate servo lifted hotends and 4 separate extruders what makes the print catridge somewhat heavy...) and that´s to much for PETG too. 

I´m interrested in drawings of
- that side mount for the nema that´s sitting just below your box driving the lead screw
- cs2-1.stl 

Could solve that encoder problem with PoKeys - if interessted here´s the description:
https://blog.poscope.com/wp-content/uploads/2017/01/DM-W16-p-57-63.pdf

kachurovskiy, 2021-05-16:
Here's the stepper mount - [nema23anglemount v9.zip](https://github.com/kachurovskiy/nanoels/files/6489970/nema23anglemount.v9.zip) - but I don't think there are any `cs2-1.stl` files in this repo - could post an URL to the image/stl file?

Charly333, 2021-05-17:
Great.  Thank you. 

The cs2-1.stl was in a zip I´ve downloaded from here. Looks like this: 
[image]

kachurovskiy, 2021-05-17:
Sure, here is the latest version of that model - [Cross Slide 2 v15.zip](https://github.com/kachurovskiy/nanoels/files/6496828/Cross.Slide.2.v15.zip)

Charly333, 2021-05-18:
Thank´s a lot. Should be easy to adapt for milling. Motor Housing needs a bit more opening and will be milled from two sides to get sharp corners wherevere needed. And front-plate - geared side - should be a bit thicker. Might have to reduce the gear thickness a bit. May hand back the adapted version after milling and testing if you like.

Is your solution of rotation readout precise enough for indexing gears? At the moment I´m producing gears by printing them with stiff transparent filament in vasemode. Filling up with glas fiber enhanced resin, it´s got more strengh than filament, but the working edges from that filament layer arn´t that brickle as if made of resin only. But even if having a 5mm clamp from a big one old style Lüsterklemme resined in for fixing gear on axis, it´s worked out after a while, specially if it´s used for bidirectional purposes...

kachurovskiy, 2021-05-18:
> Should be easy to adapt for milling.

It was designed for 3D printing, would be pretty hard to mill this if possible but let me know if it works.

> Is your solution of rotation readout precise enough for indexing gears?

Not sure what you mean. There's always backlash between gears so calculating the cutter position by stepper turns is only valid as long as direction is unchanged and stepper didn't lose steps.

There's also an option to order metal gears from China, they're inexpensive.

Another option suggested in a YouTube comment was to use chevron (herringbone) gears instead of standard ones. There are plugins for generating them both in Fusion and in FreeCAD. I didn't try them though and gear thickness might be a problem.

If you want to have no backlash, you'd need to switch from lead screws to ball screws and belt drive or direct drive.

closed 2021-06-11 by kachurovskiy