kachurovskiy/nanoels #138: Nano ELS

jvojvodi, 2023-11-10:
Hi
I put all the components together and it seems to work. But as a beginner I have a problem with setting up hardware and software. Which lines of code should be aligned and to which values and how to set the DIP switches on the stepper driver. The stepper is directly connected to the leading spindle with a pitch of 2.5 mm. Equipment specification attached.

Thanking you in advance.

[Equipment for ELS.pdf](https://github.com/kachurovskiy/nanoels/files/13319161/Equipment.for.ELS.pdf)

kachurovskiy, 2023-11-11:
Hi @jvojvodi,

1. Spindle direction: show angle on screen using top-right button. Rotate the chuck forward manually - angle should increase. If it decreases, swap encoder signal wires in the terminal.
2. Motor direction: try LEFT and RIGHT buttons - if motor is moving in the wrong direction, change SW5 DIP switch position
3. Looks like your version of driver CL57T V3.0 doesn't allow setting 200 steps resolution via DIP switches according to https://www.omc-stepperonline.com/download/CL57T.pdf so you'd either have to buy a V4.0 version of this driver or figure out if it's possible to use software to configure 200 steps - https://www.omc-stepperonline.com/closed-loop-stepper-driver-0-8-0a-24-48vdc-for-nema-17-23-24-stepper-motor-cl57t - NanoELS H2 can work with 800 steps but you will be limited in terms of motor speed you can achieve.
4. Is your encoder directly connected to spindle 1:1? In this case line 6 in h2.ino already uses `#define ENCODER_STEPS 600.0` and doesn't need to be changed.
5. Line 7 says `#define MOTOR_STEPS 200.0` - if you end up using 800 steps, change that to `#define MOTOR_STEPS 800.0`
6. Line 8 says `#define LEAD_SCREW_DU 20000.0` - your leadscrew pitch is 2.5mm so change it to `#define LEAD_SCREW_DU 25000.0`

jvojvodi, 2023-11-13:
Yes, now I understand where my problem is.

The encoder is in a ratio of 2:1 (48 teeth shaft : 24 teeth encoder). does that mean line 6 in h2 should be 300?

I will try with DIP 800, if it doesn't work, substitute for V4.0

In any case, I am very grateful for the reply

Greeting

kachurovskiy, 2023-11-13:
> The encoder is in a ratio of 2:1 (48 teeth shaft : 24 teeth encoder). does that mean line 6 in h2 should be 300?

For one turn of the shaft you'd have 2x impulses generated so use `#define ENCODER_STEPS 1200.0`. You can then enable displaying of angle on screen and check than turning the spindle by hand 1 turn results in angle changing by 360 degrees to it's start value.

Good luck!

jvojvodi, 2023-11-13:
Tx

closed 2023-11-13 by jvojvodi