kachurovskiy/nanoels #139: NanoELS - try first

jvojvodi, 2023-11-14:
I need help
Everything is connected. (see pictures) The encoder makes a full circle on the display and shows an angle of 360°. Correct. 
The problem is that the stepper does not start.
No matter what I do, the stepper stops.
Where am I wrong.

[Greeting

[image]
[image]
[image]
[image]

jvojvodi, 2023-11-14:
The video is not ideal, the display is not visible in wide view. But maybe this one will help.
The third and fourth lines of the display disappear when the encoder is turned. Is that normal?
Also, the stepper heats up a lot in this experiment.

https://github.com/kachurovskiy/nanoels/assets/150443240/8ba8e45f-0c49-4ddd-8657-60c17f61a30d

kachurovskiy, 2023-11-14:
Bottom lines disappearing during motion is intended since Nano only has 1 core and can either drive the motor or update the screen.

Look for the 24/5v switch on the side of your driver, ensure it's in the 5v position.

jvojvodi, 2023-11-15:
5V works wonders:) Everything works fine on the test bench. Now comes the implementation on my DIY lathe.
Thanks again
Greeting

closed 2023-11-15 by jvojvodi