kachurovskiy/nanoels #166: Fitting to WM180

Colinsp, 2024-01-04:
I am planning on fitting this to the WM180 lathe. The cross slide is identical so there is no issue there. However, the lathe spindle has a gear with only 40 teeth for the encoder. As a result neither this or the alternative gear wheels are suitable. Do you have suggestions for a suitable alternative please?

kachurovskiy, 2024-01-04:
Hi @Colinsp, can you please post some pictures of the relevant pieces of your lathe? Dimensions would also be helpful 👍

Colinsp, 2024-01-05:
The photo shows the headstock end of my lathe with the change gears removed so you can see more. I hope this helps.

kachurovskiy, 2024-01-05:
Got it. Can you give me A and B dimensions?

[image]

Colinsp, 2024-01-05:
I am not sure what you mean  by A from your drawing. If it is the diameter of the 40 tooth gear it is 42mm.

B is 23mm.
Thanks Colin

kachurovskiy, 2024-01-05:
Just recalled that this was already solved before in https://github.com/kachurovskiy/nanoels/discussions/159 - let's follow-up there.

closed 2024-01-06 by kachurovskiy