kachurovskiy/nanoels #168: [Feature Request] Support thread cutting in Gcode

Festivejelly, 2024-01-08:
I've been using the Nano ELS via the gcode function more and more and I think it would be nice to support threading operations through it.

Since we cant set the speed of the spindle in gcode we could instead read the RPM and adjust the feedrate as needed.

So something like G32 Z15.0 would need to work out a base feedrate and adjust that based on the RPM. I guess you did something similar with the automatic threading function on the controller.

Ive made quite a bit of progress on the Ui im building so it would be nice to have this function there.

Festivejelly, 2025-03-02:
Ive implemented this in my fork you can see an example here:

https://github.com/Festivejelly/nanoels/blob/main/h4/h4.ino#L3387C1-L3387C40

Im currently still developing it and its not fully tested yet. But it essentially uses the built in threading function for the controller. But allows us to set the values using gcode.

// G33 Threading command handler
// Format: G33 Z[end] X[end] P[pitch] H[passes] Q[start_z] R[start_x]
// Parameters:
//   Z - End Z position
//   X - End X position
//   P - Thread pitch in mm
//   H - Number of passes (optional, default 3)
//   Q - Start Z position (optional, defaults to current position)
//   R - Start X position (optional, defaults to current position)
// Example: G33 Z20 X-4.188 P1.5 H4 Q0 R-5 - External M10x1.5 thread
// Example: G33 Z20 X-5.1 P1.5 H4 Q0 R-4.5 - Internal M10x1.5 thread

kachurovskiy, 2025-03-02:
That's pretty hacky but if it works it works 🤷