kachurovskiy/nanoels #169: Gcode tool with not accounted for

KyrreKnive, 2024-01-09:
STOCK D6
TOOL ROUND R0.1 L1.5

L3 DS0 DE6 CONV

L3 DS6 DE0 CONV
L2 D3
L3 DS3 DE6

[image]

kachurovskiy, 2024-01-09:
You've specified round tool with a very small radius, did you mean to use a RECT tool?

KyrreKnive, 2024-01-09:
Yes, I am ssorry thats my fault.

kachurovskiy, 2024-01-09:
No worries, I need to make it show an error when L is specified for a round tool and when tool is smaller than the pass width. Also showing the tool in UI would help.

KyrreKnive, 2024-01-09:
I am using excel to tweek the final gcode. Changes in the editor is not vizible in grafic display?
[image]

kachurovskiy, 2024-01-09:
Wow, that's very smart 🤓 You can preview any gcode on sites like ncviewer.com

KyrreKnive, 2024-01-09:
Yes I just paste the code in column A and calculate absolute coordinates in columns E(Z) and F(X)

closed 2024-01-09 by KyrreKnive

Festivejelly, 2024-01-09:
@KyrreKnive this tool might help you https://festivejelly.github.io/LatheGcodeVisualiser/ I made it to help me tweak my gcode. Supports absolute and relative co ordinates.