kachurovskiy/nanoels #170: Speed display

duebelpapst, 2024-01-10:
Hello Maxim,

I have noticed that sometimes 0 rpm is displayed for the speed, but when I switch to angle, the angle is displayed correctly and the thread cutting also works.
So it can't be the encoder, do you have an explanation?

Regards,
Stefan

kachurovskiy, 2024-01-10:
Hi Stefan, RPM calculation averages over multiple readings and requires a short period of time to get established when you turn on the RPM display.

We only calculate RPM when it's needed - https://github.com/kachurovskiy/nanoels/blob/4e57cae627a4846d0f2991b280499fa76635bc59/h4/h4.ino#L3018 - to save processing time since it runs on the main thread which also runs the steppers.

If you want, you can change that condition to always calculate it - this way there will be no delay when turned on.

duebelpapst, 2024-01-10:
Thank's, that's what I guess, but I just wanted to ask to be sure.

Regards,
Stefan