kachurovskiy/nanoels #183: Facing and Cut-off issues

lorandzt, 2024-02-28:
Hi Maxim,
The X axis when is controlled by the rotation of the spindle ( facing or cutoff) moves the cross-slide about 2/3 of what should per spindle rotation. The return movement, moving the cross-slide with the arrows is OK. No issues with Z axis, turning works well.

For example, if I do facing in 1 steps on 20mm diameter, the cross-slide (X axis) moves 6.29mm instead of 10mm then returns 10mm. 
On the X axis I use the DM556Y motor driver and is set for 800 steps. I have the 1:3 belt drive connection to the cross-slide screw that has 1mm pitch.

Not sure if there is a correlation error between the encoder and the X axis movement or I'm
doing something wrong.
Let me know what you think.
Thank you,
Lorand

kachurovskiy, 2024-02-28:
Hi @lorandzt, sorry to hear that you're facing this issue. If it would help if you could share a video of the problem and maybe a few photos of your build.

1. Do you get insufficient X axis movement when using manual move buttons as well?
2. Please try swapping X and Z terminals on the back of H4. Does the problem now happen on Z axis or remains on X?
3. Please try reducing start speed and acceleration on X axis by 30% and see if it helps.

Thanks!

lorandzt, 2024-02-28:
Hi Maxim,
Weirdly, using manual movement from the buttons was working well for all axis.
I swapped the X and Z terminals and it worked ok so I replaced the DM556Y motor controller with a DM556 and now is working just fine.
Originally I bought two DM556Y for this project and it turned out that both are malfunctioning. I had to replace the original Z axis driver because the stepper was constantly stalling.
I attached couple of pictures. 

Thanks for your help, 
Lorand

[image]
[image]

closed 2024-03-01 by lorandzt