kachurovskiy/nanoels #191: some doubts regarding the project

serlancelot, 2024-04-23:
hello!  I have seen your ELS project and I have some doubts...  and I would like if you can answer them before I start buying the components.

The first is the use of a 4-layer PCB, is this for any specific reason?  (signal shielding?) Can only two layers be used?

The second is if you have thought about or are going to put a color LCD screen with graphics, in that case could the same PCB be used or would it be new?

Thank you very much for creating this project!!

kachurovskiy, 2024-04-23:
I'm using 4 layers because at large quantities the cost difference is marginal and signal shielding is noticeably better giving better chances during regulatory compliance tests.

It's a good idea to go with just 2 layer for just a small personal run. You could also get rid of a lot of optional components like filters, fuses, ferrites etc.

I've thought about using a more elaborate LCD but that would make the project MUCH more complicated. Currently the entire thing fits in 1 file, 3k LoC. It's easy to understand and change. Fancy LCD logic will require 3rd party software and much more file/code complexity.

Thanks!

mfxpyro1, 2024-06-23:
You can also remove the 5v to 3v3 converters as the ESP32 inputs are fully 5v tolerant.

kachurovskiy, 2024-06-24:
> You can also remove the 5v to 3v3 converters as the ESP32 inputs are fully 5v tolerant and pretty much any 5v logic input will work fine with 3V3.

It's not 5V tolerant, https://www.espressif.com/sites/default/files/documentation/esp32-s3-wroom-1_wroom-1u_datasheet_en.pdf page 17

It's likely that ESP32 inputs won't immediately fry with 5V inputs (encoder) but we still need to output 5V on most signal pins to drive the stepper motor drivers which for the most parts require 5V to work.

mfxpyro1, 2024-06-24:
> It's not 5V tolerant, https://www.espressif.com/sites/default/files/documentation/esp32-s3-wroom-1_wroom-1u_datasheet_en.pdf page 17
>
> It's likely that ESP32 inputs won't immediately fry with 5V [...]

The CEO of Espressif has gone on record to say despite what that document says the pins ARE 5v tolerant only the power isn't, I've regularly used 3v3 for step/direction driver inputs as they're virtually also opto isolated and the LED's have a reasonable wide input range.

https://voodoo.business/2021/05/19/are-the-esp32-and-esp8266-5v-tolerant-yes-they-officially-are/

"When asked why this information is not in the datasheet, he responded
“the reason is too many users took it to mean that the chip is 5 V tolerant. When we say 5 V tolerant, we are only referring to the IOs. So some users mistook this to make that they can power the chip entirely off the 5 V supply. The correct usage is to use 5 V open for these 5 V tolerant pins, and only via only drain configuration.” And then elaborated on the matter with “I understand, but the time needed to do the iterations when mistakes were made, was too long. when the product was launched 5 V WiFi modules (with DCDC) were the norm. Many users saw “5 V” written in the specs and thought it could be a 1-1 replacement for such modules.”

mfxpyro1, 2024-06-26:
> hello! I have seen your ELS project and I have some doubts... and I would like if you can answer them before I start buying the components.
>
> The first is the use of a 4-layer PCB, is this for any [...]

I've just built one on prototyping board (Vero/Perf board) including the keypad so about as basic as you can get, it uses an off the shelf ESP32 dev board module and an Adafruit module for the key scanning, case and keypad will be fully 3D printed, not as pretty as the original design but a lot cheaper and just as functional, as I mentioned in my other comment I left out the 5v-3v3 converters as IME they are unnecessary in 99.9% of cases I also left out most of the filtering etc. parts which are needed for a commercial product but not strictly needed for a hobby build..

kachurovskiy, 2024-06-26:
@mfxpyro1 very nice, please kindly post any relevant photos and schematics here for others to use if you can. Thanks 🙏

mfxpyro1, 2024-06-27:
Thanks I will but it'll take a little while as I'm redoing the keypad for larger buttons which also means a new case.

mfxpyro1, 2024-07-03:
> @mfxpyro1 very nice, please kindly post any relevant photos and schematics here for others to use if you can. Thanks 🙏

Here you go, hope it works.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

RGUERI1, 2025-02-21:
Very well done.

demsarmatjaz, 2025-02-22:
Great job @mfxpyro1. I wonder how did you implement I2C keypad? That is exactly what I am looking for.

mfxpyro1, 2025-02-22:
Same as the original but IC is pre soldered onto a breakout board, it was
bought from Amazon, keypad was "bodge wired" on matrix board and everything
else 3D printed.

demsarmatjaz, 2025-02-23:
Thanks for the answer. I probably phrased the question a bit incorrectly... I know you used the Adafruit circuit with the TCA8418 encoder, the question was more in terms of how you modified the code to read the keyboard via the I2C protocol, not the serial one, which the original code uses?

mfxpyro1, 2025-02-23:
I didn't modify the code it works exactly the same way as the original.

demsarmatjaz, 2025-02-23:
ah, of course, how stupid I am... this is the H4 version and not the H5. The H4 basically works with the TCA8418.

richoffpints, 2025-04-14:
did this h4 work?

richoffpints, 2025-04-14:
@mfxpyro1