kachurovskiy/nanoels #192: RPM gets stuck sometimes

digiexchris, 2024-04-24:
Hi, I'm not quite sure under what circumstances this happens, but sometimes my RPM display gets stuck. It shows 0 rpm when stopped, and when running it goes to the last set RPM, even if I change rpm. It stops updating. I need to power cycle the system to get it to respond again.

Is this a known issue?

kachurovskiy, 2024-04-24:
Hi Chris, seems like a new issue.

I guess the condition at https://github.com/kachurovskiy/nanoels/blob/main/h4/h4.ino#L580 is not overflow-proof.

Try replacing the first 2 conditions in that function with the following and let me know if that helps?

```
unsigned long t = micros();
unsigned long elapsedTime = t - spindleEncTime;
if (elapsedTime > 50000) {
  // RPM less than 10.
  spindleEncTimeDiffBulk = 0;
  return 0;
}
if (elapsedTime < RPM_UPDATE_INTERVAL_MICROS) {
  // Don't update RPM too often to avoid flickering.
  return shownRpm;
}
```

digiexchris, 2024-04-26:
sure I'll give that a try. I'm planning on putting a different encoder on in a few days, I'll do it then.

vegetate7, 2024-05-02:
Hello! Maxim, I'll try you fixes tomorrow. But. I think it may be not only RPM display issue. I saw same time axis moving stopped. Either automatic and manual (showing "SYNC" on display). Like it losing a sync to spindle encoder. But same time angle display still working correctly.

kachurovskiy, 2024-05-02:
It's possible for the lead screw to go out of sync with the spindle by removing the limit while the carriage is standing on it. It's indicated by the SYN word on the display.

This situation should resolve itself once the spindle makes a full turn, mode changes or the stop button is pressed.

vegetate7, 2024-05-02:
> @vegetate7 it's possible for the lead screw to go out of sync with the spindle by removing the limit while the carriage is standing on it. It's indicated by the SYN word on the display.
> 
> This situation should resolve itself once the spindle makes a full turn, mode changes or the stop button is pressed.

But it is not this case. When RPM stops, i can not move carriage by arrow buttons. The stepper start humming, and no moving (I have "need rest" option enabled). This state only can be disabled by powercycle. Or by pressing "stop" button, and powercycle after it because "Stop in manual mode".
Auto modes can not be started too. Again, stepper start hummin, display shows "step 0", carriage not moving.
Twise i have this issue in the middle of turn pass: carriage stopped and no moving after.
The spindle keep rotating in all those cases.
So IMO something wrong in the "processSpindleCounter()". I planned to look closer, but had no time yet.

kachurovskiy, 2024-05-03:
@vegetate7 that sounds like a separate issue we should look into. Please file a new issue with the steps to reproduce if you can

vegetate7, 2024-05-03:
> @vegetate7 that sounds like a separate issue we should look into. Please file a new issue with the steps to reproduce if you can

Ok, I'll do it later.
WRT proposed changes: It shold be ok, I do not see why not. But it completely stops to count RPM. Suppose it should be something very obviouse :) Can not look closer, It's machining right now..

vegetate7, 2024-05-03:
> Ok, I'll do it later. WRT proposed changes: It shold be ok, I do not see why not. But it completely stops to count RPM. Suppose it should be something very obviouse :) Can not look closer, It's [...]

ah.. I think it's here:
`
...
  if (elapsedTime < RPM_UPDATE_INTERVAL_MICROS) {
...
`
'elapsedTime' should be rebased to 'showRpmTime' first:
`
....
  unsigned long elapsedTime = t - showRpmTime;
  if (elapsedTime < RPM_UPDATE_INTERVAL_MICROS) {
...
`

kachurovskiy, 2026-01-01:
This has long been fixed in H5.