kachurovskiy/nanoels #194: Carriage stops moving

vegetate7, 2024-05-04:
Sometimes the carriage stops moving. Auto modes no starting (shows "step 0"). Manual moves not starting too. It just start humming by stepper, but no move. I'm using "need rest" option, so steppers disabled at idle.
Hard to tell how to reproduce. Like make some moves, and then keep system powered on for long time (1hr 11 min, i guess :) without steppers activity).
I think it's here it not overflow-proof. Not sure, tho, is it only my branch affected, or the main too.
in the `void moveAxis(Axis* a)`
my branch:
```
...
  unsigned long nowUs = micros();
  float delayUs = 1000000.0 / a->speed;
  if ((nowUs - a->stepStartUs < delayUs - 5)||
      ((a->posByDRO && a->backlashByDRO && a->needsDROBacklashCompensation) &&
       (a->stepStartUs + DRO_READ_PERIOD_MS * 1000 > nowUs)) ){
    // Not enough time has passed to issue this step.
    return;
  }
 ... 
```
main:
```
...
  unsigned long nowUs = micros();
  float delayUs = 1000000.0 / a->speed;
  if (nowUs - a->stepStartUs < delayUs - 5) {
    // Not enough time has passed to issue this step.
    return;
  }
...
```

micros() rotating aroung 1hr 11min, so if last 'stepStartUs' taken too long ago, it can became significantly bigger then 'nowUs'.
I suppose both branches may be affected, but I'm not very good in the overflow handle programmin. If it only mine, I'll be glad to have some advices how to fix.

kachurovskiy, 2024-05-04:
Hi @vegetate7, thanks for reporting this.

1. Can you please take a photo of the screen next time this happens?
2. Can you please attach your h4.ino with your next comment or share it via https://gist.github.com/?
3. I'm not aware of the way to get into the `step 0` state - it doesn't accept 0 values - do you know how to do that?

Thanks!

vegetate7, 2024-05-04:
My h4.ino: https://github.com/vegetate7/nanoels_dro/blob/main/h4/h4.ino
Just looked at the code, not the 'step 0', but actually 'Pass 0 of ...', I just wrongly remembered, sorry :)
Not going to machining right now, so can not take photo.
But I had a time to play with code. I thing it's a bit of "silent type conversion" magic, and this:
```
  if ((nowUs - a->stepStartUs < delayUs - 5)||
      ((a->posByDRO && a->backlashByDRO && a->needsDROBacklashCompensation) &&
       (a->stepStartUs + DRO_READ_PERIOD_MS * 1000 > nowUs)) ){
    // Not enough time has passed to issue this step.
    return;
  }
```
should be done as this:
```
  if ((nowUs - a->stepStartUs < delayUs - 5)||
      ((a->posByDRO && a->backlashByDRO && a->needsDROBacklashCompensation) &&
       (nowUs - a->stepStartUs < DRO_READ_PERIOD_MS * 1000)) ){
    // Not enough time has passed to issue this step.
    return;
  }
```
Your code is already OK here. IMO. I'll try it a bit later.

kachurovskiy, 2024-05-04:
Could it be that you switched to spindle reverse and forgot to switch back? This is the usual cause of automatic operations not starting.

vegetate7, 2024-05-04:
Nope. I usually starting operations when spindle already running. I'll test my fix in few days, just not right now.

vegetate7, 2024-06-29:
Well, I had a chance to test. Yes, type conversion trick fixes issue. I'll close this one.

closed 2024-06-29 by vegetate7