kachurovskiy/nanoels #199: error: too many arguments to function 'hw_timer_t*

atmelletje, 2024-06-01:
hello Maxim,

I have some issues getting the ESP32 programmed.
IDE 2.3.2 
updated and installed the ESP32 boards including the Adafruit lib.
I'm fairly new to programming such devices. would be nice if you could push me in the right direction.

h4\h4.ino:570:37: error: too many arguments to function 'hw_timer_t* timerBegin(uint32_t)'
  570 | hw_timer_t *async_timer = timerBegin(0, 80, true);
      |                           ~~~~~~~~~~^~~~~~~~~~~~~
 'hw_timer_t* timerBegin(uint32_t)'

Roel

kachurovskiy, 2024-06-01:
Hi @atmelletje, not sure, try changing your esp32 library version to 2.0? https://github.com/espressif/arduino-esp32/issues/8765#issuecomment-1867399333

atmelletje, 2024-06-02:
hello Maxim, first of all, thank you for your fast reply.
what I did is the following workaround, 
I went back to Arduino v1.8.18 and installed the  Espressif esp32 v2.0.17
That worked like a Champ.

Danke Maxim,

Roel

jherbert7621, 2024-06-20:
Hi @atmelletje,
had the same issue, as Maxim suggested changing esp32 library to 2.0 does the trick!
Juergen

atmelletje, 2024-06-21:
it works great, I have some boards left over. they are completely assembled and working. drop me a message if someone needs one! a complete set is also possible. next to replace are the leadscrews with some decent ballscrew spindles.

kachurovskiy, 2024-08-29:
This should now be resolved thanks to https://github.com/kachurovskiy/nanoels/commit/f0e5ebeef747257782caa141d6bcb181ce0b972a

closed 2024-08-29 by kachurovskiy