kachurovskiy/nanoels #245: H5 Compile Error

AlanMerry, 2025-01-12:
I was attempting to upload the h5.ino file to the controller but got this error:-

C:\Users\alann\OneDrive\Documents\3DPrinterProjects\ELSBuild\nanoels-main\h5\h5.ino: In function 'void setup()':
C:\Users\alann\OneDrive\Documents\3DPrinterProjects\ELSBuild\nanoels-main\h5\h5.ino:1425:13: error: 'A12' was not declared in this scope; did you mean 'A2'?
 1425 |     pinMode(A12, OUTPUT);
      |             ^~~
      |             A2
C:\Users\alann\OneDrive\Documents\3DPrinterProjects\ELSBuild\nanoels-main\h5\h5.ino:1426:13: error: 'A13' was not declared in this scope; did you mean 'A3'?
 1426 |     pinMode(A13, OUTPUT);
      |             ^~~
      |             A3
C:\Users\alann\OneDrive\Documents\3DPrinterProjects\ELSBuild\nanoels-main\h5\h5.ino:1427:13: error: 'A11' was not declared in this scope; did you mean 'B11'?
 1427 |     pinMode(A11, OUTPUT);
      |             ^~~
      |             B11
C:\Users\alann\OneDrive\Documents\3DPrinterProjects\ELSBuild\nanoels-main\h5\h5.ino:1440:179: error: 'A11' was not declared in this scope; did you mean 'B11'?
 1440 |   initAxis(&y, NAME_Y, ACTIVE_Y, ROTARY_Y, MOTOR_STEPS_Y, SCREW_Y_DU, SPEED_START_Y, SPEED_MANUAL_MOVE_Y, ACCELERATION_Y, INVERT_Y, NEEDS_REST_Y, MAX_TRAVEL_MM_Y, BACKLASH_DU_Y, A11, A12, A13, Y_PULSE_A, Y_PULSE_B, PCNT_UNIT_3);
      |                                                                                                                                                                                   ^~~
      |                                                                                                                                                                                   B11
C:\Users\alann\OneDrive\Documents\3DPrinterProjects\ELSBuild\nanoels-main\h5\h5.ino:1440:184: error: 'A12' was not declared in this scope; did you mean 'A2'?
 1440 |   initAxis(&y, NAME_Y, ACTIVE_Y, ROTARY_Y, MOTOR_STEPS_Y, SCREW_Y_DU, SPEED_START_Y, SPEED_MANUAL_MOVE_Y, ACCELERATION_Y, INVERT_Y, NEEDS_REST_Y, MAX_TRAVEL_MM_Y, BACKLASH_DU_Y, A11, A12, A13, Y_PULSE_A, Y_PULSE_B, PCNT_UNIT_3);
      |                                                                                                                                                                                        ^~~
      |                                                                                                                                                                                        A2
C:\Users\alann\OneDrive\Documents\3DPrinterProjects\ELSBuild\nanoels-main\h5\h5.ino:1440:189: error: 'A13' was not declared in this scope; did you mean 'A3'?
 1440 |   initAxis(&y, NAME_Y, ACTIVE_Y, ROTARY_Y, MOTOR_STEPS_Y, SCREW_Y_DU, SPEED_START_Y, SPEED_MANUAL_MOVE_Y, ACCELERATION_Y, INVERT_Y, NEEDS_REST_Y, MAX_TRAVEL_MM_Y, BACKLASH_DU_Y, A11, A12, A13, Y_PULSE_A, Y_PULSE_B, PCNT_UNIT_3);
      |                                                                                                                                                                                             ^~~
      |                                                                                                                                                                                             A3

exit status 1

Compilation error: 'A12' was not declared in this scope; did you mean 'A2'?

kachurovskiy, 2025-01-12:
@AlanMerry weird how it even compiles for most of us. Fixed in https://github.com/kachurovskiy/nanoels/commit/65d622a9ad9edec2793c18dfa00d603a79a59bb0 - thanks for reporting.

AlanMerry, 2025-01-12:
Thank you Maxim, it has now compiled successfully and I have now uploaded to the controller.

closed 2025-01-13 by kachurovskiy