kachurovskiy/nanoels #283: Movement rough via Gcode but smooth via controller

Festivejelly, 2025-03-02:
I'm noticing that when moving the axis using the directional buttons on the Z the movement is nice and smooth.

But when sending fast commands by gcode its quite jerky. Even though technically the feedrate is lower.

I suspect this is the culprit:

long chunks = round(max(max(abs(xDiff), abs(zDiff)), abs(a1Diff)) * LINEAR_INTERPOLATION_PRECISION);
for (long i = 0; i < chunks; i++) {
    if (!isOn) return;
    float scale = i / float(chunks);
    stepToContinuous(&x, xStart + xDiff * scale);
    stepToContinuous(&z, zStart + zDiff * scale);
    if (ACTIVE_A1) stepToContinuous(&a1, a1Start + a1Diff * scale);
    gcodeWaitNear();
}

I didnt notice it before when I was running much slower speeds. But now im running with ballscrews im pushing it a bit faster.

I think maybe using velocity planning rather than chunking would result in a smoother move and probably a better finish.

I'll have a crack at implementing this, just wanted to raise the issue to track it but also see if there any alternative solutions.

kachurovskiy, 2025-03-02:
Can you give an example GCode where the movement is jerky and share video of the problem?

Festivejelly, 2025-03-02:
If I send G1 Z30 F800 its a bit jerky.

Anything up to 500 seems to be fine.

Ill take a video next time im in my workshop. You cant really see it jerking, you can just feel it and hear it.

kachurovskiy, 2025-04-14:
> the movement remains jerky

Is it jerky at the start, end, or all the way through at equal intervals?

> Maybe I can tweak these values?

Sure, give it a try

Festivejelly, 2025-04-15:
> > the movement remains jerky
> 
> Is it jerky at the start, end, or all the way through at equal intervals?
> 
> > Maybe I can tweak these values?
> 
> Sure, give it a try

Its jerky the entire movement.

So lets say I do Z10 F800 its very jerky, or maybe more accurately it kinda vibrates.

The faster the feedrate the more kind vibrates.

I cant get near the jog speed of the manual controls without it vibrating too much.

kachurovskiy, 2025-04-15:
I think this is caused by using high number of steps per revolution and `GCODE_WAIT_EPSILON_STEPS` being just a constant 10. Try increasing it to e.g. 50 and seeing if it helps.

Festivejelly, 2025-04-15:
> I think this is caused by using high number of steps per revolution and `GCODE_WAIT_EPSILON_STEPS` being just a constant 10. Try increasing it to e.g. 50 and seeing if it helps.

Ill try that.

What would be the downside of not moving in chunks? Its not clear why the movements are done in chunks for gcode. Is it so we can send stop commands?

For example:

// Rapid positioning / linear interpolation.
void G00_01(const String& command) {
  long xStart = x.pos;
  long zStart = z.pos;
  long a1Start = a1.pos;
  long xEnd = command.indexOf(x.name) >= 0 ? mmOrInchToAbsolutePos(&x, getFloat(command, x.name)) : xStart;
  long zEnd = command.indexOf(z.name) >= 0 ? mmOrInchToAbsolutePos(&z, getFloat(command, z.name)) : zStart;
  long a1End = command.indexOf(a1.name) >= 0 ? mmOrInchToAbsolutePos(&a1, getFloat(command, a1.name)) : a1Start;
  long xDiff = xEnd - xStart;
  long zDiff = zEnd - zStart;
  long a1Diff = a1End - a1Start;

  // Calculate required speeds for coordinated movement
  updateAxisSpeeds(xDiff, zDiff, a1Diff);

  // Command axes to move directly to the final position
  // The underlying stepTo/moveAxis logic should handle smooth motion
  stepToFinal(&x, xEnd);
  stepToFinal(&z, zEnd);
  if (ACTIVE_A1) stepToFinal(&a1, a1End);

  // Wait for the entire move to complete
  gcodeWaitStop();
}

Festivejelly, 2025-04-15:
> Yes, IIRC the main reason is being able to stop when the spindle stops or a stop command is issued via keyboard or other means.

So could we do something like this instead?

void gcodeWaitEpsilon(int epsilon) {
  while ((abs(x.pendingPos) > epsilon || 
          abs(z.pendingPos) > epsilon || 
          abs(a1.pendingPos) > epsilon || 
          (SPINDLE_PAUSES_GCODE && getApproxRpm() < GCODE_MIN_RPM)) && isOn) {
    taskYIELD();
  }
  
  // If motion was cancelled (isOn is false either from the controller button or via sending the the stop command by gcode)
  if (!isOn) {
    if (xSemaphoreTake(motionMutex, 100) == pdTRUE) {
      x.pendingPos = 0;
      z.pendingPos = 0;
      if (ACTIVE_A1) a1.pendingPos = 0;
      xSemaphoreGive(motionMutex);
    }
  }
}

kachurovskiy, 2025-04-15:
I would not advise that, try playing with `GCODE_WAIT_EPSILON_STEPS`.

Festivejelly, 2025-04-16:
> I would not advise that, try playing with `GCODE_WAIT_EPSILON_STEPS`.

I did make the change... because well... im a cowboy haha. What are the risks of me using my code? It seems to still react to stop commands.