kachurovskiy/nanoels #285: H4 - real stepmotor  off function.

RobinI30N, 2025-03-09:
I made my own compilation ver. H4. I changed the LCD to I2C and the QWERTY I2C keyboard.
https://docs.m5stack.com/en/unit/cardkb_1.1

[image]

I had a problem with manual movement from the keyboard: if I move the axis it moves infinitely.
I noticed that the keyboard handling thread during movement is not handled - the key release code is not read. The solution was - I don't know if it's correct? - but it works:

void taskDisplay(void *param) {
  while (emergencyStop == ESTOP_NONE) {
    updateDisplay();
    processKeypadEvent();   <------------------------

I also want to turn off the engine (release the ENA signal): if I turn off a given axis on the display, it will allow manual movement of the lathe wheels.
Could you give me some advice on how to achieve this?

kachurovskiy, 2025-03-09:
Manual movement and keyboard handling is indeed on the same core (https://github.com/kachurovskiy/nanoels/blob/main/h4/h4.ino#L1754-L1757) but e.g. taskMoveZ() yields by calling `waitForPendingPosNear0()`. I'm not sure why calling `processKeypadEvent()` from another thread on the same core would make a difference.

You can auto-disable axes when they aren't moving via `NEEDS_REST_Z` and `NEEDS_REST_X` configuration params or use the respective buttons to control this yourself - https://github.com/kachurovskiy/nanoels/blob/main/h4/h4.ino#L2568-L2573

RobinI30N, 2025-03-09:
Thanks for the tip.
1. Could it be a problem that the LCD and keyboard are the same I2C bus?

2. There is a function in the code that does this.

void updateEnable(Axis* a) {
  if (!a->disabled && (!a->needsRest || a->stepperEnableCounter > 0)) {
    DHIGH(a->ena);
    // Stepper driver needs some time before it will react to pulses.
    DELAY(STEPPED_ENABLE_DELAY_MS);
  } else {
    DLOW(a->ena);
  }
}

GPIO state is changing, now I know I have an electrical problem with the controller

Thank you

RobinI30N, 2025-03-15:
appreciate the amount of work you put into the software.

However, when all the code is in one file, analysis is very difficult.

I am looking for a place in the code to change the movement distance per one movement encoder pulse.

It would be natural for one pulse to move the axis by 0.01, instead of 0.015.

The wheel has 100/ppr.

kachurovskiy, 2025-03-15:
Imagine how difficult it would be for you with 25 files in 4 folders :) With 1 file you can at least search or scroll.

Just increase `PULSE_PER_REVOLUTION` in https://github.com/kachurovskiy/nanoels/blob/main/h4/h4.ino#L65C13-L65C34 by 33%

RobinI30N, 2025-03-15:
With a 100 pulse wheel, I had PULSE_PER_REVOLUTION = 100 entered.
After changing to PULSE_PER_REVOLUTION = 150 I have 0.01mm/p - I would never have discovered that.

For now I did it for the Z axis, I don't know how it will affect the X axis :)

RobinI30N, 2025-04-04:
I re-ran the changes with custom LCD and keyboard customization on a clean h4v12 file and the keyboard thread problem disappeared. I didn't check what was wrong before.

closed 2025-04-04 by RobinI30N