kachurovskiy/nanoels #288: web interface to control the H5 system's operation？

ningjiazun, 2025-03-11:
Have you considered implementing a web interface to control the H5 system's operation, similar to how you currently manage G-code via a web interface? This would allow you to leverage existing devices (e.g., smartphones, tablets, or computers) as control screens, eliminating the need for dedicated hardware. You could reference Duet3D's DuetWebControl (a web-based control system for 3D printers) for design inspiration and functional implementation.

kotovasia5120, 2025-03-11:
Yes of course this is in the plans.

closed 2025-03-11 by kotovasia5120

kachurovskiy, 2025-03-11:
@kotovasia5120 would it make sense to keep this issue open for tracking purposes?

kotovasia5120, 2025-03-11:
> @kotovasia5120 would it make sense to keep this issue open for tracking purposes?

🤷‍♂️In such cases, they usually use a division into tags, issues, and feature requests. Decide for yourself.

kachurovskiy, 2025-03-11:
@ningjiazun 3D printer is a bit of a different machine, I run my printer without even looking at it - just send the job over. Stock is always loaded, machine has end stops and auto bed leveling.

With lathe, it's not really possible to do anything remotely - at least with machines NanoEls is aimed at, you have to be at the machine to change the tool, insert the stock, zero the tool, watch it run to adjust the RPM or stop.

Eliminating a 22 EUR keyboard to replace it with a 100 EUR smartphone or tablet that is less good at the job (not tactile) seems strange. Or do you suggest to eliminate the Nextion screen too?

arvin2009, 2025-03-11:
اگرر در این نمایشگر کلید ها طراحی شود خوب  است و گر نه برای نمایش چند خط بسیار گران است

ningjiazun, 2025-03-14:
Implement a web-based control interface to connect the ESP32 board via the local Wi-Fi network, enabling control and monitoring through existing electronic devices. This setup also facilitates seamless OTA  updates in the future.

ningjiazun, 2025-03-14:
Additionally, using the Nextion serial touch screen solely for status display without leveraging its interactive control capabilities seems like underutilization of resources。