kachurovskiy/nanoels #297: Compilation error: too few arguments to function 'hw_timer_t* timerBegin(uint8_t, uint16_t, bool)'

arvin2009, 2025-03-15:
What is the cause of this error in H5?
I searched and it said to download the library platform-espressif32-51.03.03 but it didn't work.

arvin2009, 2025-03-15:
Look here. I did this and the software completely froze and got stuck.
[https://community.platformio.org/t/problem-compiling-code-for-esp32-dev-module/41873](url)

kachurovskiy, 2025-03-16:
Most likely a version problem with one of the libraries of board.

Try backing up your sketches, uninstalling Arduino and installing everything again as per https://github.com/kachurovskiy/nanoels/tree/main/h5#programming-the-controller

arvin2009, 2025-03-16:
I uninstalled and reinstalled several times. The error persists. The problem is here.esp32 boards 2.0.18 
The non-valid version 3 must be used. But I couldn't install that either.

arvin2009, 2025-03-16:
What version of IDE are you using?

kachurovskiy, 2025-03-16:
Arduino IDE 2.3.4, esp boards 3.1.3. Looks like you have old version of the esp32 board definitions installed, needs to be updated to 3.x.x.

[image]

arvin2009, 2025-03-16:
Exactly right. I had the old one installed and didn't see this one. I think it's fixed. Thanks.

closed 2025-03-20 by kachurovskiy