kachurovskiy/nanoels #304: left and right limits is reversed ?

RobinI30N, 2025-04-02:
Last version H4V12.

Hello
I am at the stage of starting and checking.

I have the impression that the meaning of the left and right limits is reversed.
If the symbol >| is displayed, it suggests setting the right limit. The action is such that the carriage stops at the limit moving to the left.

|< works the other way around. Up / down limits works according to the symbol.

The second inconsistency is in Face mode - "left to right" processes the left surface and should take the material moving to the left.

I understand that the X axis has positive values ​​on the right side - they increase when I move away from the spindle?

kachurovskiy, 2025-04-02:
The long axis on the lathe is usually called Z. In our case we assume that it increments to the left, towards the spindle. Sounds like you need to invert the Z axis motor in the code or on the driver.

RobinI30N, 2025-04-02:
[image]
Let's systematize, I assumed according to the drawing.
Is there any inconsistency with the NanoELS convention?

RobinI30N, 2025-04-04:
Yes.. that was the cause.
Changing the direction of the Z axis solved this problem..also other unreported ones.

It doesn't matter to me anymore but you could include a drawing of the valid axis directions in the documentation.

Thank you and best regards.

closed 2025-04-04 by RobinI30N

RobinI30N, 2025-04-04:
".. if motor is moving in the wrong direction..,"
Only it is not known which direction is correct - for the author it is obvious.

You can adjust the wheel, keyboard and engine so that the manual movement is consistent, but for the program it is incorrect (my case).

I tried to solve this without assuming inssues from your video and it came out as it did - i.e. the other way around. ;)

RobinI30N, 2025-04-04:
yes.. because the entire project except for the software was mine. Changing the type of keyboard, LCD, schematic and PCB.

I managed to set everything up so that everything worked properly except for the limits.

I immediately ruled out a software error but I hit a wall and ran out of ideas.

That's why I asked on github.