kachurovskiy/nanoels #308: h4

richoffpints, 2025-04-09:
hey so this is my first ime buying and building a pcb from online, i have the folder with the gerber and the bom and pcbs files, can you please give me a step by step on how tob ordder all the components for the h4 boards. i feel like the way im doing it is wrong, i dont wana spend money and cuzim dumb and miss somethung. this is such a great project. litterly iv looked and your makingcrazycoolshit i love it

kachurovskiy, 2025-04-09:
Hi @richoffpints, it's very hard to build H4 because it's a complex PCB and a lot of components specified are no longer available. It's also expensive.

I suggest you build H5 instead, it's much cheaper and easier. Process of ordering the pcbs is described in the 5th version documentation.

richoffpints, 2025-04-09:
can you tell me how i can convert the h5 into a 4axis? is it possible even?

richoffpints, 2025-04-14:
so how do i change the code to use z scale output on  a new axis? is it easy? i know c so u dont need to skid it