kachurovskiy/nanoels #324: PCNT driver error

slapukas, 2025-05-18:
При компиляции выводит сообщение:
#warning "legacy pcnt driver is deprecated, please migrate to use driver/pulse_cnt.h"
при старте 
19:54:04.862 -> E (2899) pcnt(legacy): _pcnt_get_counter_value(108): PCNT driver error
19:54:04.896 -> E (2899) pcnt(legacy): _pcnt_get_counter_value(108): PCNT driver error
19:54:04.896 -> E (2899) pcnt(legacy): _pcnt_get_counter_value(108): PCNT driver error
как это устранить????

Arduino IDE 2.3.6 ESP32 v. 3.20

kachurovskiy, 2025-05-20:
Странно, какая у вас версия ESP32? Нужно ESP32-S3 например N16R8.

slapukas, 2025-05-21:
версия N16R8, странно то, что и h4 тоже выдает при старте PCNT driwer error. С ESP IDF 2.17 pcnt.h при компиляции проходит без преупреждения о миграции но  PCNT driver error все равно. Максим, если можно, какие у Вас установки в arduino ide для ESP32S3 N16R8? ИИ (ChatGPT) бредит о возмошности конфликта пинов, нету ни киких конфлитов.

kachurovskiy, 2025-05-21:
Попробуйте отключить дисплей при загрузке прошивки. Насколько я понимаю ваш дисплей сам имеет внутри еще одну ESP32 и подключён к тому же серийному каналу.

slapukas, 2025-05-21:
Эта белеберда выдается и без дисплея, но..... энкодеры работают! С такой конфигурацией PCNT ошибки не дает :

#include "driver/pcnt.h"

#define PCNT_H_LIM    310000
#define PCNT_L_LIM    300000
#define PCNT_FILTER   1  // 1 APB clock cycles (~1 us)

typedef struct {
  pcnt_unit_t unit;
  int gpioA;
  int gpioB;
} EncoderConfig;

EncoderConfig encoders[4] = {
  { PCNT_UNIT_0, 13, 14 },
  { PCNT_UNIT_1, 18, 8 },
  { PCNT_UNIT_2, 47, 21 },
  { PCNT_UNIT_3, 45, 48 }
};

void setupEncoder(const EncoderConfig& config) {
  pcnt_config_t pcntConfig = {};
  pcntConfig.pulse_gpio_num = config.gpioA;
  pcntConfig.ctrl_gpio_num = config.gpioB;
  pcntConfig.channel = PCNT_CHANNEL_0;
  pcntConfig.unit = config.unit;
  pcntConfig.pos_mode = PCNT_COUNT_INC;     // Count on A rising
  pcntConfig.neg_mode = PCNT_COUNT_DEC;     // Count on A falling
  pcntConfig.lctrl_mode = PCNT_MODE_REVERSE;
  pcntConfig.hctrl_mode = PCNT_MODE_KEEP;
  pcntConfig.counter_h_lim = PCNT_H_LIM;
  pcntConfig.counter_l_lim = PCNT_L_LIM;

  pcnt_unit_config(&pcntConfig);
  pcnt_set_filter_value(config.unit, PCNT_FILTER);
  pcnt_filter_enable(config.unit);
  pcnt_counter_pause(config.unit);
  pcnt_counter_clear(config.unit);
  pcnt_counter_resume(config.unit);
}

void setup() {
  Serial.begin(115200);
  delay(1000);
  Serial.println("Starting 3 rotary encoders using PCNT...");

  for (int i = 0; i < 4; i++) {
    setupEncoder(encoders[i]);
  }
}

void loop() {
  for (int i = 0; i < 4; i++) {
    int16_t count = 0;
    esp_err_t err = pcnt_get_counter_value(encoders[i].unit, &count);
    if (err == ESP_OK) {
      Serial.printf("Encoder %d count: %d\n", i + 1, count);
    } else {
      Serial.printf("Encoder %d read error: %d\n", i + 1, err);
    }
  }
  Serial.println("--------------------------");
  delay(1000);
}

kachurovskiy, 2025-05-25:
[image]

Мне кажется это происходит потому что мы пытаемся прочитать счетчики перед тем как они инициализированы. Попробуйте переставить строку 3335 на 3331 и скажите помогает или нет.

closed 2025-05-25 by kachurovskiy