kachurovskiy/nanoels #325: Loss steps

Jjjigu, 2025-05-22:
Thank you so much for sharing the project, it's amazing.
 I've been using the H2 for a year and I'm very happy with it, but I've decided to switch to the H5.  Nema 24 in Z and 17 in X .

I've encountered a problem. 
In manual movements ,and in displacement movements without material loading or when threading at high speed,  there is a huge loss of steps that prevents the axes from moving, and the H5 makes changes to the selected function, steps, etc., at the moment this step loss occurs. It started happening with the X-axis, but now the Z-axis as well—it seems like the ESP is not capable of sending so many pulses. 

The configuration is: X-axis 800 steps in the driver and 2400 in H5, and Z-axis 800 in the driver and 600 in H5.

I have changed the travel speed in SPEED_MANUAL_MOVE_Z = 8 * MOTOR_STEPS_Z. In the X-axis, I replaced the 8 with 3, and in the Z-axis, I changed the 8 to 5. It also does this in the movements that the H5 performs during automatic displacement without material loading or threading at 400 RPM and a pitch of 1.25mm. 

What values should I adjust to prevent this? Is there any other speed adjustment? Errors continue to occur in the H5 when this happens because it changes values, functions...  

By lowering the previous speed, it seems to happen much less, but at some point, it happens again.

On the other hand, when generating G-code, the X-axis movements make a loud noise similar to a step loss, but it doesn’t seem to actually lose steps.

Thank you very much for your help.

kachurovskiy, 2025-05-22:
This usually happens on brand-less cheap stepper drivers. Can you please post some pictures of your setup including wiring and all the components involved? Also your entire h5.ino via https://gist.github.com/. Thanks!

Jjjigu, 2025-05-22:
Hi, attach the pictures

[image]

this the stepper Z nema 34

[image]

[image]

driver Z ()this is good driver) (800 microsteps)

[image]

stepper X nema 17 1:3

[image]

driver X TB6600 (400 microsteps)

[image]

[image]

H5

[image]

With the configuration I've attached, all the H5's functions work well except for G-Code, which constantly loses steps only on the Z-axis. Could it be a problem with the TB6600?

kachurovskiy, 2025-05-23:
Yes, brand less cheap drivers like TB6600 give very rough movement and are likely to lose steps. Try a quality driver from e.g. Stepperonline or Rtelligent.

demsarmatjaz, 2025-05-23:
Hi, I would preffer closed loop steppers and appropriate drivers. As Maxim told, those drivers are not the best choice. The stepper itself is strong enough with 8,5Nm (if I am not wrong) but it must be driven with quality driver to get this torque from it. What voltage power supply did you take for Z axix driver? Maybe you could try with higher voltage, the driver can accept from 24VDC to 80VDC, I would go with at least 60VDC but rather 72VDC. Just my opinion...

Jjjigu, 2025-05-23:
The **Z-axis** is not giving me any problems with that driver, which is of good quality, and the power supply is **60 volts**. I only have issues if the speed increases significantly, but that doesn't worry me because I usually don't work at that speed. 

The main problem is with the Z-axis, where I have a TB6600 driver with a 20-volt, 4-amp power supply, and the motor is 2 amps, and with this configirarion  only en gcode .

https://gist.github.com/Jjjigu/132e7efb54d0253a84019fc1700facf4

Do you recommend a specific driver? I was considering the **DM-556**.

Jjjigu, 2025-05-23:
Hi, i have test this configuration and the first test run ok in G code

const long SCREW_X_DU = 20000; // Husillo de bolas SFU1204 de 4 mm en deci-micrones (10^-7 de un metro)
const long MOTOR_STEPS_X = 2400;
const long SPEED_START_X = 400; // Velocidad inicial del motor, pasos / segundo.     MOTOR_STEPS_X
const long ACCELERATION_X = 4 * MOTOR_STEPS_X; // Aceleración del motor, pasos / segundo^2.
const long SPEED_MANUAL_MOVE_X = 3000; // Velocidad máxima del motor en movimiento manual, pasos / segundo.  2* MOTOR_STEPS_X
const bool INVERT_X = true; // Cambiar (true/false) si el carro se mueve p. ej. "a la izquierda" cuando presionas "derecha".
const bool NEEDS_REST_X = false; // Falso para cualquier tipo de driver o el eje X se desbloqueará cuando no esté en movimiento.
const long MAX_TRAVEL_MM_X = 100; // El carro transversal no permite viajar más de esto de una sola vez, 10 cm.
const long BACKLASH_DU_X = 0.05; // 0.15 mm de holgura en deci-micrones (10^-7 de un metro)
const char NAME_X = 'X'; // Texto mostrado en pantalla antes del valor de la posición del eje, nombre del eje en GCode.