kachurovskiy/nanoels #327: RMT to step generation.

RobinI30N, 2025-05-25:
What do you think.. would changing the step generation method to hardware RMT have any positive effects?

I am currently working on generating movement using this method.

kachurovskiy, 2025-05-25:
Sorry, not sure what RMT is and what problem are you looking to address.

RobinI30N, 2025-05-25:
I want to introduce a new quality :)
Generally speaking, RMT is a hardware square wave generator, it allows you to generate a series of pulses with a previously set timing for each pulse to control a stepper motor.
In this application, you can prepare the entire movement, including ramps. Unlike software pulse generation, it provides more precise movement.
The hardware handles the sending, relieving the load on the processor. Many CNC controllers use this method - for example (FluidNC).
If you want to continue developing the project, you should consider it, because you will eventually reach the end of the processor's computing capabilities.
When I have a demo ready, I can attach it for testing.

kachurovskiy, 2025-05-25:
It used to be a problem with Arduino but with ESP32 we are no longer constrained on pulse generation unless you go with very high micro-stepping.

Currently we know exactly how many pulses were issued and have 0 accumulating error. With hardware timers IIUC it's not possible to know how many pulses are issued exactly. The problem with accumulating error in timers is well demonstrated on YouTube with Chipmaker ELS.

Happy to be proven wrong and to see an improvement upon the existing work.