kachurovskiy/nanoels #342: Positional Error on Z axis

Festivejelly, 2025-08-16:
I've got a Z axis set up with these params:

const long SCREW_Z_DU = 50000; // 5mm lead screw in deci-microns (10^-7 of a meter)
const long MOTOR_STEPS_Z = 2000;
const long SPEED_START_Z = 2 * MOTOR_STEPS_Z; // Initial speed of a motor, steps / second.
const long ACCELERATION_Z = 20 * MOTOR_STEPS_Z; // Acceleration of a motor, steps / second ^ 2.
const long SPEED_MANUAL_MOVE_Z = 6 * MOTOR_STEPS_Z; // Maximum speed of a motor during manual move, steps / second.
const bool INVERT_Z = true; // change (true/false) if the carriage moves e.g. "left" when you press "right".
const bool NEEDS_REST_Z = false; // Set to false for closed-loop drivers, true for open-loop.
const long MAX_TRAVEL_MM_Z = 300; // Lathe bed doesn't allow to travel more than this in one go, 30cm / ~1 foot
const long BACKLASH_DU_Z = 200; // 400 = 0.04mm backlash in deci-microns (10^-7 of a meter)
const char NAME_Z = 'Z'; // Text shown on screen before axis position value, GCode axis name

I notice that when moving using the 1mm step the controller will often move short or too far by the backlash value.

For example if I zero Z and move left, then move right I end up with Z being 0.02mm short. So for example instead of the position being Z4 it ends up being Z3.98

This value changes based on the backlash amount set. So if I change the backlash to 400 du then the Z shows up as Z3.96

Any ideas what might be causing this? I put some logging in:

Move #1 - Before: z.pos=0, z.motorPos=0, z.originPos=0, z.pendingPos=0, z.fractionalPos=0.0000, moveStep=10000, direction=LEFT, z.backlashSteps=8
Move #1 - After: z.pos=4800, z.motorPos=4800, z.originPos=0, z.pendingPos=0, z.fractionalPos=0.0000, z.backlashSteps=8
Move #2 - Before: z.pos=4800, z.motorPos=4800, z.originPos=0, z.pendingPos=0, z.fractionalPos=0.0000, moveStep=10000, direction=RIGHT, z.backlashSteps=8
Move #2 - After: z.pos=1192, z.motorPos=1184, z.originPos=0, z.pendingPos=0, z.fractionalPos=0.0000, z.backlashSteps=8
Move #3 - Before: z.pos=1192, z.motorPos=1184, z.originPos=0, z.pendingPos=0, z.fractionalPos=0.0000, moveStep=10000, direction=RIGHT, z.backlashSteps=8
Move #3 - After: z.pos=-808, z.motorPos=-816, z.originPos=0, z.pendingPos=0, z.fractionalPos=0.0000, z.backlashSteps=8
Move #4 - Before: z.pos=-808, z.motorPos=-816, z.originPos=0, z.pendingPos=0, z.fractionalPos=0.0000, moveStep=10000, direction=RIGHT, z.backlashSteps=8
Move #4 - After: z.pos=-2008, z.motorPos=-2016, z.originPos=0, z.pendingPos=0, z.fractionalPos=0.0000, z.backlashSteps=8
Move #5 - Before: z.pos=-2008, z.motorPos=-2016, z.originPos=0, z.pendingPos=0, z.fractionalPos=0.0000, moveStep=10000, direction=LEFT, z.backlashSteps=8
Move #5 - After: z.pos=400, z.motorPos=400, z.originPos=0, z.pendingPos=0, z.fractionalPos=0.0000, z.backlashSteps=8

If I move via gcode I dont seem to get this positional error. Similarly if I change my Z axis pitch to 2mm instead of 5 then it also goes away.

This is on a H4 btw

Festivejelly, 2025-08-16:
With a 2mm pitch:

Move #1 - Before: z.pos=0, z.motorPos=0, z.originPos=0, z.pendingPos=0, z.fractionalPos=0.0000, moveStep=10000, direction=LEFT, z.backlashSteps=20
Move #1 - After: z.pos=9000, z.motorPos=9000, z.originPos=0, z.pendingPos=0, z.fractionalPos=0.0000, z.backlashSteps=20
Move #2 - Before: z.pos=9000, z.motorPos=9000, z.originPos=0, z.pendingPos=0, z.fractionalPos=0.0000, moveStep=10000, direction=RIGHT, z.backlashSteps=20
Move #2 - After: z.pos=-5000, z.motorPos=-5020, z.originPos=0, z.pendingPos=0, z.fractionalPos=0.0000, z.backlashSteps=20

I've tried to see where the differences are but my head is hurting now lol

kachurovskiy, 2025-08-17:
Both GCode and manual moves use `stepToContinuous()` so there should not be a difference in backlash logic since it's implemented inside - https://github.com/kachurovskiy/nanoels/blob/main/h4/h4.ino#L2617

Perhaps the error comes from the fact that on the initial move we assume the backlash is taken on the right side of the tool.

Festivejelly, 2025-08-19:
Weirdly adding DELAY(50); in the taskMoveZ seemed to fix it...

        z.speedMax = getStepMaxSpeed(&z);
        stepToContinuous(&z, posCopy + delta);
        waitForStep(&z);
        DELAY(50);
      } while (delta != 0 && (left ? buttonLeftPressed : buttonRightPressed));

Festivejelly, 2025-08-19:
Im not sure if that would be the cause, even if i wait like 10 seconds then do the change in direction it still appears. It seems to be tied to the pitch of the leadscrew and steps. My x axis is fine and doesnt exhibit this problem with a 2mm pitch.

But the 5mm pitch on z seems to cause issues.

I suspected it was because the task was being run over and over and not completing in time. Hence why i added a delay, and the delay seemed to fix it.